RADIUSdesk/rdcore #1: NAS on the latest version

dd1245, 2022-01-25:
Hi.

On the latest version of RADIUSdesk, I can create RADIUS clients, but do not seem to be able to create a NAS now, with its own key. Is this still possible, and if not, If I manually insert an entry into the SQL nas table, will this still work?

antoniovalenzuela, 2022-04-04:
I can't add NAS Direct (fixed IP). 

Add "+" only can add NAS Dynamic. 

RD OVA VirtualBox.

dirkvanderwalt, 2022-04-04:
Hi,

I need to check if the code paw ported to CakePHPv3.
If so, its probably just commented out and we can add maybe a config file setting so it can be displayed.

Nudge me again if I have not replied in 24 hours :-)

antoniovalenzuela, 2022-04-04:
Thanks.
I need direct NAS to setup POD/COA 3799, because currently I can't kill user connected to Coova.

antoniovalenzuela, 2022-04-05:
Please add timezone "America/Chile" now GMT-4 and DST with GMT-3. 

I don't undestand the values of the table "timezone".

I think it would be better to read the timezones of the O.S.

;)

antoniovalenzuela, 2022-04-05:
Hi @dirkvanderwalt 

could you check the code?

antoniovalenzuela, 2022-04-05:
> Please add timezone "America/Chile" now GMT-4 and DST with GMT-3.
> 
> I don't undestand the values of the table "timezone".
> 
> I think it would be better to read the timezones of the O.S.
> 
> ;)

Solved, I added into DB. GMT-4.

antoniovalenzuela, 2022-04-06:
@dirkvanderwalt ?

dirkvanderwalt, 2022-04-08:
Hi everyone,

I checked the code and it is there 

https://github.com/RADIUSdesk/rdcore/tree/main/rd/classic/src/view/nas
https://github.com/RADIUSdesk/rdcore/blob/main/cake3/rd_cake/src/Controller/NasController.php

So I should have a way to enable / include it checked in (into Github) by Monday.

dd1245, 2022-04-08:
> Hi everyone,
> 
> I checked the code and it is there
> 
> https://github.com/RADIUSdesk/rdcore/tree/main/rd/classic/src/view/nas https://github.com/RADIUSdesk/rdcore/blob/main/cake3/rd_cake/src/Controller/NasController.php
> 
> So I should have a way to enable / include it checked in (into Github) by Monday.

That's great. Thanks

dirkvanderwalt, 2022-04-09:
Actually you should be able to activate them even in the current code 
Just remove the comments (for root user)
https://github.com/RADIUSdesk/rdcore/blob/main/cake3/rd_cake/src/Controller/DashboardController.php#L795

And then for the other normal Access Providers
https://github.com/RADIUSdesk/rdcore/blob/main/cake3/rd_cake/src/Controller/DashboardController.php#L1513

Let us know if this works for you

dd1245, 2022-04-09:
Seems to work fine, thanks very much.

antoniovalenzuela, 2022-04-10:
@dirkvanderwalt  
Works, but some issues:

- Can't delete added "NAS Devices"
- Kick doesn't work for CoovaChilli COA port 3799 (data saved in optional info)

To kick or disconnect I added in public function "kickActive" file RadacctsController.php

shell_exec("echo User-Name='".$ent->username."' | radclient -c '1' -n '3' -r '3' -t '3' -x '127.0.0.1:3799' 'disconnect' 'testing123'");

dedykurniadi, 2022-06-09:
> @dirkvanderwalt Works, but some issues:
>
> * Can't delete added "NAS Devices"
> * Kick doesn't work for CoovaChilli COA port 3799 (data saved in optional info)
>
> To kick or disconnect I added in public [...]

Can you give me example? I've add line like this 

shell_exec("echo User-Name='".$ent->username."' | radclient -c '1' -n '3' -r '3' -t '3' -x '".$ent->
nasipaddress.:3799"' 'disconnect' 'secret'");

but I still cant kick the user, I use .$ent->nasipaddress. to determine which NAS, because  I have many Mikrotik NAS.

dedykurniadi, 2022-06-13:
can you help @antoniovalenzuela

dedykurniadi, 2022-06-13:
> shell_exec("echo User-Name='".$ent->username."' | radclient -c '1' -n '3' -r '3' -t '3' -x '".$ent-> nasipaddress.:3799"' 'disconnect' 'secret'");

ther right is 

shell_exec("echo User-Name='".$ent->username."' | radclient -c '1' -n '3' -r '3' -t '3' -x '".$ent->
nasipaddress.":3799' 'disconnect' 'secret'");

for now i can kick the client

netcom-rs, 2022-11-23:
Hi.

Is there a way to dynamically pull secret from database for radclient? It is useful if we have different secrets for each client.

Best regards,
Aleksa.