RADIUSdesk/rdcore #2: authenticate.json is missed

forthgate, 2022-02-20:
I can't log in to radiusdesk dashboard, looks like some directories and scripts are missed.  I get the error 404 cake3/rd_cake/dashboard/authenticate.json in the browser console.

dirkvanderwalt, 2022-05-21:
Can you try again. and let us know.
We added a lot of fixes and new features since this report.

bile0026, 2022-11-16:
Similar issue here on ubuntu 22.04 LTS, following the new instructions for ubuntu 24.04 with php8.1.  The web interface will not load, I get the loading (scrolling blue bars), and then a blank white page. The "branding.json" file doesn't exist.

```
"GET /cake4/rd_cake/dashboard/branding.json?_dc=1668567950233&sel_language=4_4 HTTP/1.1" 500 405
```