RADIUSdesk/rdcore #26: How to upgrade from RADIUSdesk-2022-B0.ova to latest github code?

razametal, 2022-12-29:
Hi,

I want upgrade from the version RADIUSdesk-2022-B0.ova to the latest code on github. There is any documentation about make it possible?

Any help will be appreciated.

Tempokgs, 2023-01-04:
https://www.radiusdesk.com/docuwiki/getting_started/update

razametal, 2023-07-20:
Hi, thank you for the respones.

Yes, the client traffic report tab is working on this version

I'm trying to update, but the path of the application is a little different from the version Version 21-01-5 that I have installed.

Latest github version have the option to register customers filling a form an then using accessing to the internet, I want to have these feature on mi installation.

Maybe it is possible to copy the the realms and permanent users data to a new installation of the actual version?