RADIUSdesk/rdcore #36: Ajax communication failed (error 500) only for wizard and cloud

bolemo, 2023-04-20:
Greetings,

As I want to find a solution to monitor a running and working freeradius setup as a small ISP lab point of view, I gave a try to RADIUSdesk.

I installed a fresh copy of RADIUSdesk on an ubuntu 22.04 VM following these instructions: https://www.radiusdesk.com/wiki/getting_started/22_install_ubuntu_nginx#install_radiusdesk

I made a few changes, as I have a nginx reverse proxy for it, and it points to `/var/www/radiusdesk/` instead of `/var/www/html/`, and I made the symlinks to `radiusdesk` instead of `html`, and modified the paths properly from the tutorial, and for the cron4 tasks as well.

I did not sync with the running freeradius yet, as I encounter what seems to be a bug.
I can access the pages, PHP is working fine, I can change the settings, admins…
But I encounter an error as soon as I touch the Clouds, either from the wizard or the OTHERS->Clouds tab.

Creating a cloud (there is of course none by default) is giving me an `Ajax communication failed` on the GUI (`Failed to load resource: the server responded with a status of 500 ()` on `https://radiusdesk.mydomain.com/cake4/rd_cake/wizards/new-site-step-one.json` or `https://radiusdesk.mydomain.com/cake4/rd_cake/clouds/add.json`)

The log shows this:
```
2023-04-17 11:36:13 warning: DebugKit is disabling itself as your host `radiusdesk.mydomain.com` is not in the known safe list of top-level-domains (localhost, invalid, test, example, local). If you would like to force DebugKit on use the `DebugKit.forceEnable` Configure option.
2023-04-17 11:36:13 error: [PDOException] SQLSTATE[HY000]: General error: 1364 Field 'description' doesn't have a default value in /var/www/rdcore/cake4/rd_cake/vendor/cakephp/cakephp/src/Database/Statement/MysqlStatement.php on line 39
Stack Trace:
- /var/www/rdcore/cake4/rd_cake/vendor/cakephp/cakephp/src/Database/Statement/MysqlStatement.php:39
- /var/www/rdcore/cake4/rd_cake/vendor/cakephp/cakephp/src/Database/Connection.php:360
- /var/www/rdcore/cake4/rd_cake/vendor/cakephp/cakephp/src/Core/Retry/CommandRetry.php:70
- /var/www/rdcore/cake4/rd_cake/vendor/cakephp/cakephp/src/Database/Connection.php:363
- /var/www/rdcore/cake4/rd_cake/vendor/cakephp/cakephp/src/Database/Query.php:254
- /var/www/rdcore/cake4/rd_cake/vendor/cakephp/cakephp/src/ORM/Table.php:2091
- /var/www/rdcore/cake4/rd_cake/vendor/cakephp/cakephp/src/ORM/Table.php:1984
- /var/www/rdcore/cake4/rd_cake/vendor/cakephp/cakephp/src/ORM/Table.php:1883
- /var/www/rdcore/cake4/rd_cake/vendor/cakephp/cakephp/src/ORM/Table.php:1570
- /var/www/rdcore/cake4/rd_cake/vendor/cakephp/cakephp/src/Database/Connection.php:709
- /var/www/rdcore/cake4/rd_cake/vendor/cakephp/cakephp/src/ORM/Table.php:1571
- /var/www/rdcore/cake4/rd_cake/vendor/cakephp/cakephp/src/ORM/Table.php:1884
- /var/www/rdcore/cake4/rd_cake/src/Controller/WizardsController.php:765
- /var/www/rdcore/cake4/rd_cake/src/Controller/WizardsController.php:176
- /var/www/rdcore/cake4/rd_cake/vendor/cakephp/cakephp/src/Controller/Controller.php:547
- /var/www/rdcore/cake4/rd_cake/vendor/cakephp/cakephp/src/Controller/ControllerFactory.php:139
- /var/www/rdcore/cake4/rd_cake/vendor/cakephp/cakephp/src/Controller/ControllerFactory.php:114
- /var/www/rdcore/cake4/rd_cake/vendor/cakephp/cakephp/src/Http/BaseApplication.php:320
- /var/www/rdcore/cake4/rd_cake/vendor/cakephp/cakephp/src/Http/Runner.php:77
- /var/www/rdcore/cake4/rd_cake/vendor/cakephp/cakephp/src/Http/Middleware/BodyParserMiddleware.php:162
- /var/www/rdcore/cake4/rd_cake/vendor/cakephp/cakephp/src/Http/Runner.php:73
- /var/www/rdcore/cake4/rd_cake/vendor/cakephp/cakephp/src/Routing/Middleware/RoutingMiddleware.php:186
- /var/www/rdcore/cake4/rd_cake/vendor/cakephp/cakephp/src/Http/Runner.php:73
- /var/www/rdcore/cake4/rd_cake/vendor/cakephp/cakephp/src/Routing/Middleware/AssetMiddleware.php:77
- /var/www/rdcore/cake4/rd_cake/vendor/cakephp/cakephp/src/Http/Runner.php:73
- /var/www/rdcore/cake4/rd_cake/vendor/cakephp/cakephp/src/Error/Middleware/ErrorHandlerMiddleware.php:131
- /var/www/rdcore/cake4/rd_cake/vendor/cakephp/cakephp/src/Http/Runner.php:73
- /var/www/rdcore/cake4/rd_cake/vendor/cakephp/cakephp/src/Http/Runner.php:58
- /var/www/rdcore/cake4/rd_cake/vendor/cakephp/cakephp/src/Http/Server.php:90
- /var/www/rdcore/cake4/rd_cake/webroot/index.php:40
- /var/www/rdcore/cake4/rd_cake/index.php:16

Request URL: /wizards/new-site-step-one.json
Referer URL: https://radiusdesk.mydomain.com/rd/build/production/Rd/
Client IP: xxxx:xxxx:xxxx:xxxx:xxxx:xxxx:xxxx:xxxx
2023-04-19 11:34:52 warning: DebugKit is disabling itself as your host `radiusdesk.mydomain.com` is not in the known safe list of top-level-domains (localhost, invalid, test, example, local). If you would like to force DebugKit on use the `DebugKit.forceEnable` Configure option.
```

The problem is likely not nginx or php related, as all the rest works well, but anyway, here is my nginx config in `/etc/nginx/site-enabled/radiusdesk.mydomain.com`:
```
server {
    listen              80;
    listen              [::]:80;

    server_name         radiusdesk.mydomain.com;

    include             snippets/certbot.conf;

    location / {
	return              301 https://radiusdesk.mydomain.com$request_uri;
    }
}

server {
    listen	443 ssl http2;
    listen	[::]:443 ssl http2;
    server_name	radiusdesk.mydomain.com;

    allow	xxxx:xxxx:xxxx::/48;
    deny	all;

    add_header	Strict-Transport-Security "max-age=31536000; includeSubDomains" always;

    ssl_certificate	/path/to/certs/radiusdesk.mydomain.com/fullchain.pem;
    ssl_certificate_key	/path/to/certs/certs/radiusdesk.mydomain.com/privkey.pem;

    access_log	/var/log/nginx/access_radiusdesk.log;

    proxy_set_header	Host $http_host;
    include	snippets/robots.conf;
    include	snippets/certbot.conf;

    root /var/www/radiusdesk;
    index index.php index.html index.htm;

    # pass PHP scripts to FastCGI server
    #
    location ~ \.php$ {
      include snippets/fastcgi-php.conf;
      #
      #       # With php-fpm (or other unix sockets):
      fastcgi_pass unix:/var/run/php/php8.1-fpm.sock;
      #       # With php-cgi (or other tcp sockets):
      #       fastcgi_pass 127.0.0.1:9000;
    }

    # deny access to .htaccess files, if Apache's document root
    # concurs with nginx's one
    #
    location ~ /\.ht {
      deny all;
    }

    location /cake4/rd_cake/node-reports/submit_report.json {
      try_files $uri $uri/ /reporting/reporting.php;
    }

    location /cake4/rd_cake {
      rewrite ^/cake4/rd_cake(.+)$ /cake4/rd_cake/webroot$1 break;
      try_files $uri $uri/ /cake4/rd_cake/index.php$is_args$args;
    }

    location ~ ^/cake4/.+\.(jpg|jpeg|gif|png|ico|js|css)$ {
      rewrite ^/cake4/rd_cake/webroot/(.*)$ /cake4/rd_cake/webroot/$1 break;
      rewrite ^/cake4/rd_cake/(.*)$ /cake4/rd_cake/webroot/$1 break;
      access_log off;
      expires max;
      add_header Cache-Control public;
    }

    location / {
      try_files $uri $uri/ =404;
    }
}
```

Thank you for you assistance :)

predators46, 2023-05-05:
@bolemo 

how do you install radiusdesk in ubuntu? did you run composer

bolemo, 2023-05-05:
> @bolemo
> 
> how do you install radiusdesk in ubuntu? did you run composer

No, I did not use Docker at all.

I followed the instructions here: https://www.radiusdesk.com/wiki/getting_started/22_install_ubuntu_nginx
Except nginx and freeradius were already installed.

My goal is strictly monitoring the existing freeradius (used in an ISP logic to authorize customer devices on DHCP from their router's MAC); I don't need the AP / captive portal features…

dkay-zaxxx, 2024-01-25:
hello did you come right? also getting the same error