RADIUSdesk/rdcore #39: Doesnt Work

ValkyrX, 2023-06-18:
No matter what i do and no matter what install howto i follow when i goto the dashboard i get a loading in the center of the screen and then a blank screen, i have done this on ubuntu 20.04 22.04 and also a proxmoc container vm, none work .. Any ideas ?

umblox, 2023-07-01:
[image]
me too,

FreezyExp, 2023-07-09:
Same I did a quick test running the docker-compose image and it seems the database is not getting configured properly to allow the connection.

mfzhsn, 2023-07-24:
@FreezyExp  @ValkyrX  I am struck while creating a docker image with some errors, I am looking for some help. 

Also did your problem ?

`+ docker-compose up -d radiusdesk
/usr/lib/python3/dist-packages/paramiko/transport.py:236: CryptographyDeprecationWarning: Blowfish has been deprecated
  "class": algorithms.Blowfish,
Starting radiusdesk-mariadb ... error

ERROR: for radiusdesk-mariadb  Cannot start service rdmariadb: Mounts denied:
The path /mnt/data/radiusdesk/db_conf/my_custom.cnf is not shared from the host and is not known to Docker.
You can configure shared paths from Docker -> Preferences... -> Resources -> File Sharing.
See https://docs.docker.com/ for more info.

ERROR: for rdmariadb  Cannot start service rdmariadb: Mounts denied:
The path /mnt/data/radiusdesk/db_conf/my_custom.cnf is not shared from the host and is not known to Docker.
You can configure shared paths from Docker -> Preferences... -> Resources -> File Sharing.
See https://docs.docker.com/ for more info.
ERROR: Encountered errors while bringing up the project.`