RADIUSdesk/rdcore #45: stuck on loading

ZohaibBinKhalid, 2023-09-27:
installation completed without any bug, but showing only loading, please guide me how can i fix this
screenshort attached 
[image]

bobcat23-pixel, 2023-12-09:
I am also having the same issue and I think it is a database error or nginx config files issues. Could someone help us out

ZohaibBinKhalid, 2023-12-17:
i just upgraded php 7.5 to 8.1

nitewolf75, 2024-09-20:
i also have the same issue, followed all the instructions on the website. instillation on Ubuntu server 22.04.
if i try  the mobile URL( rd_mobile) , the login page loads, but i cannot login. nothing happens if i put in the username and password .

jatinnijash, 2024-10-03:
i am also get same issue with ubuntu 22.04 and docker also

Castrobrine, 2025-01-24:
> i also have the same issue, followed all the instructions on the website. instillation on Ubuntu server 22.04. if i try the mobile URL( rd_mobile) , the login page loads, but i cannot login. nothing happens when i put in the default username and password .

OMG I have the same problem, don't know what to do. Did you manage to find a fix?