RADIUSdesk/rdcore #53: Missing SQL tables after clean installation

diegoproquo, 2024-01-08:
Hi, i am running a clean installation of Ubuntu 22.04, with php8.1, following all the steps from the official wiki. Everything looks and work fine, except for the permanent users. When i try to add a new one, an alert pops up with the message "Ajax communication failed". Going depper into the error through the console, i get this message: "SQLSTATE[42502]: Base table or view not found: 1146 Table `rd.realm_vlans' doesn´t exist". I check through the terminal into the rd database and indeed the table does not exist. Makes no sense to me because i imported all the tables following the guide, i have about 150. Does anyone know why is this missing?

razametal, 2024-02-02:
Hi, I've the same issue with the start wizard.

XHRPOST
https://miip.ddns.net/cake4/rd_cake/wizards/new-site-step-one.json
[HTTP/1.1 500 Internal Server Error 226ms]

	
message	
url	"/wizards/new-site-step-one.json"
code	500
file	"/var/www/rdcore/cake4/rd_cake/vendor/cakephp/cakephp/src/Database/Schema/Collection.php"
line	160

razametal, 2024-02-02:
Solved,

I need to apply some patches to mysql tables:

mysql -u root rd < /var/www/rdcore/cake4/rd_cake/setup/db/8.079_add_ppsk_field.sql 
mysql -u root rd < /var/www/rdcore/cake4/rd_cake/setup/db/8.080_add_cell_density.sql 
mysql -u root rd < /var/www/rdcore/cake4/rd_cake/setup/db/8.081_update_node_neighbors.sql 

Now the installation is working fine.