RADIUSdesk/rdcore #73: loading...screen problem

jatinnijash, 2024-10-03:
i am setup radiusdesk on ubuntu 22.0.4 installing docker complete but http://44.204.187.171/rd/ this path not accessble on loading screen display and on inspect mode "Failed to load resource: the server responded with a stat
[image]
[image]
us of 404 (Not Found)" many files

88tom, 2024-10-08:
Hi, have you verified the nginx config? This part in particular:
`location ~ ^/cake4/.+\.(jpg|jpeg|gif|png|ico|js|css)$ {
    rewrite ^/cake4/rd_cake/webroot/(.*)$ /cake4/rd_cake/webroot/$1 break;
    rewrite ^/cake4/rd_cake/(.*)$ /cake4/rd_cake/webroot/$1 break;
    access_log off;
    expires max;
    add_header Cache-Control public;
}`

jatinnijash, 2024-10-08:
Any one share me nginx config file 
Full sample??? With complete working

jatinnijash, 2024-10-08:
Cam  Hi, have you verified the nginx config? This part in particular: `location ~ ^/cake4/.+\.(jpg|jpeg|gif|png|ico|js|css)$ { rewrite ^/cake4/rd_cake/webroot/(.*)$ /cake4/rd_cake/webroot/$1 break; rewrite ^/cake4/rd_cake/(.*)$ /cake4/rd_cake/webroot/$1 break; access_log off; expires max; add_header Cache-Control public; }`

Can you provide me nginx config file with working completely??? 
Please

LucaCraft89, 2024-10-11:
`ufw allow 80` did it for me now i am stuck on Error loading manifest file to check for updates

jatinnijash, 2024-10-13:
anyone get solution ??

deangoldhill, 2024-10-18:
The problem is that the frontend application is configured to use 'localhost' as the database server instead of the database container.
Edit the file: /var/www/html/cake4/rd_cake/config/app_local.php
In the datasources section, change the hostname from 'localhost' to 'rdmariadb'

then the page loads and seems to work.
The radius service itself is already setup correctly.

But the porblem I still have is the reports are not working. The 'Data Usage' page is just blank despite authentication and accounting in the activity monitor being populated.
Probably because there are some other files with the wrong hostname specified.

jatinnijash, 2024-10-20:
i am try docker base setup and i am try install and setup on ubuntu 22.04 and 24.04 also RADIUSdesk-2023-A0.ova image all radiusdesk not working same loading error on /rd

deangoldhill, 2024-10-20:
Don't use /rd
Just put the IP or hostname in the address bar.
It will redirect you to /#dashboard and the page should load

jatinnijash, 2024-10-20:
anyone know what is username and password of RADIUSdesk-2023-A0.ova terminal login??

deangoldhill, 2024-10-20:
https://www.osboxes.org/faq/what-are-the-credentials-for-virtual-machine-image/