RADIUSdesk/rdcore #85: Install from source - errors in scripts

nitinuae, 2024-12-16:
Hello
Appreciate the project - as it seems quote a lot of effort has gone to it.
I have tried it (for home) but ran into MULTIPLE issues.
First issue is (when you find time) - documentation needs to be worked on.

The follwoing errors were observed.
a) Install from source - In AP mode - the AP never finds the internet(gateway). The internet connection remains mesh. And no stats are processed by RD. The number of users remains 0, data 0 and gateway remains blank. I tested this error as switching the vmware file from my install from scratc(Ubuntu 24.04) to yours OVA. Your OVA works fine.
b) Asia/JNB is hardcoded so many places - its almost impossible to get reports.
c) I am having MANY issues in PPSK - but still testing that. VLAN on PPSK is giving lots of grief(not working - even when using the preisntalled OVA)

Meanwhile - can you please check your code uploaded to git and compare to what you have used in OVA - and find what is causing the issue(a)

Thank you

88tom, 2024-12-17:
Hi! For the point A. Did you copy the cron file to the correct location? And if it's a net install did you install cron??

nitinuae, 2024-12-17:
I cut and paste all the instructions from wiki24

I will check in few hours if from was included in those instructions

If not, will install.

I did cut paste of cron instructions (it's almost last before tweak your install)

Thanks

nitinuae, 2024-12-17:
Thank you - the problem of LAN (instead of mesh as internet connection - auto detect) is SOLVED after install of cron.
It was a ubuntu basic install and I did cut and paste of sudo apt-get as per the instructions.
Seemscron was missing in those(Basic ubuntu 24.04 = no cron)

nitinuae, 2024-12-19:
As a suggestion- please edit and add -
1. Cron to be added in sudo add package line.
2.  Ask users to run all the sql files to create tables(your doc just hints - need to be clearly stated that all database needs to be populated first)

nitinuae, 2024-12-20:
hello
Coming back here - as the radiusdesk has again reported itself as MESH and not LAN.
I have tried restarting - (both server and openwrt multiple times), but radiusdesk just doesnt want to listen to openwrt. It stays in mesh mode as connection(instead of LAN) and doesnot report any users.

What do I need to do now