RADIUSdesk/rdcore #104: Radiusdesk & Coova-chilli & & Freeradius - Authentication failure

Forrest-It, 2025-04-20:
Following the instructions on https://radiusdesk.com/wiki24/install_22_4 and https://radiusdesk.com/wiki24/install_22_4_coova I want to create a captive portal for a school wifi network. 

The captive portal also intercepts the request by redirecting to the RadiusDesk login page, but authentication fails. Analyzing the freeradius log with the 
freeradius -x command, even if the clear text password is correctly received by the same,

**(40) pl_logintime:   $RAD_CONFIG{'Cleartext-Password'} = &control:Cleartext-Password -> 'my-clear-text-password'**

in 

******0) pl_logintime:   $RAD_REQUEST{'User-Password'} = &request:User-Password -> '???fd?U G???%5?**1'****

the password appears to be heavily altered, resulting in freeradius rejecting it and i get the follow error:

(0) pap: ERROR: Cleartext password does not match "known good" password
(0) pap: Passwords don't match

I tried to use a captive portal created with OpenWrt and Coova-chilli with the same radiusdesk&freeradius: authentication occurs correctly. 

Has anyone of you had a similar problem?

Forrest-It, 2025-04-21:
Solved !

yaco89, 2025-04-24:
how you solve it?

Forrest-It, 2025-04-25:
Hi,  the password HS_UAMSECRET=change-me in coova-chilli config file must be the same that you have in uam.php file of radiusdesk