RADIUSdesk/rdcore #108: How to translate Web Gui Alert message

Forrest-It, 2025-05-18:
This is not a problem but a request for help: I ​​would like to translate the Alert messages into Italian (such as Not Available To Use On Saturday at ..etc). Can anyone explain to me how to do it? Thanks

Anera-Group, 2025-06-09:
Forrest-It, Were you able to get the alerts (email/sms) system working?

Forrest-It, 2025-06-09:
yes, normally I receive emails only for forgotten passwords and for creating new accounts. I meant to translate alert Web Gui messages like "Not Available To Use On Saturday at" and other similar messages. I don't understand if they are messages contained in the DB: I can't find any trace of this text in the program files

hasiniaina7, 2025-10-15:
"I also haven't found a way to modify the error messages sent by radreply on the captive portal, specifically those displayed right above the login or voucher area. It seems the settings are either not located within or are not permitted in the 'dynamic login page'.

[image]